NESTCollaboration/nest #26: error message during cmake: C++11 support lacking in g++, on SL6 (Scientific Linux 6) operating system

mszydagis, 2019-04-03:
I ran into a major problem on my desktop computer at work where the latest NEST repo at time of writing this (git commit 1bd3d70) refused to compile at all during testing of gcc and g++. There was a lot of misinformation or partial information on the web that did not apply to my case. I had to merge a lot of the commands and ideas from multiple sources like stack overflow dot com and Kate K. The way I was able to finally solve the problem was exactly as follows step by step (procedure very likely would have to be modified in order to apply to different flavors of Linux/ Unix systems, and/or Apple Mac of course):  CERN software repo was key solution here but yum does not work directly (homebrew for Mac?) without some pre-reqs  

wget http://linuxsoft.cern.ch/cern/slc61/x86_64/extras/RPMS/sl-release-scl-rh-1-2.slc6.noarch.rpm
sudo rpm -Uvh sl-release-scl-rh-1-2.slc6.noarch.rpm
wget http://linuxsoft.cern.ch/cern/slc62/i386/yum/extras/sl-release-scl-1-2.slc6.noarch.rpm
sudo rpm -Uvh sl-release-scl-1-2.slc6.noarch.rpm
sudo yum install sl-release-scl
sudo yum install devtoolset-7
scl enable devtoolset-7 bash

Version 6 should also work, but I believe this is bleeding-edge latest at time of writing. Note above how downloading and installing stuff not enough: final step is to *enable*. Not sure how general this is to everybody, but perhaps the README needs to updated in next release, to make sure everyone is aware of the minimum version pre-reqs on g++ (and clang for Mac) at the very least. rootNEST also no longer compiles the old way with gcc given in the README, but works fine via cmake now if you turn ROOT support on with the proper switch.  But again, commands above will have slightly different names by OS, so I'm unsure what to say in the README without being too ultra-specific to my own personal case, or what to add to Travis.

tunnell, 2019-07-30:
Not sure what to do here.  This is just specifying the gcc and cmake versions, which is in readme, right?

jecutter, 2019-07-30:
I agree with Chris, that making sure the machine has the necessary compiler
pre-reqs should be up to the user, and is too case specific. We provide the
optional flags for specifying the compiler, using ROOT, etc.

mszydagis, 2019-07-30:
OK, I will close. This was just me in a bad mood one day.

closed 2019-07-30 by mszydagis