NESTCollaboration/nest #32: double-colon definition of "beta" particles

mszydagis, 2019-07-31:
NEST::beta and INTERACTION_TYPE::beta seem to be two solutions to the same problem of CentOS 7 disambiguation. LZ did it one way, but LUX another. Which one is right?? @tunnell @kreczko @riffard @jpbrodsky @jecutter ??

kreczko, 2019-08-01:
Given the current implementation, `NEST::beta` is the correct one (the aim is to provide a unique identifier - `INTERACTION_TYPE` does not denote any origin).

However, the better implementation would be to elevate `INTERACTION_TYPE` to an enum class (see https://en.cppreference.com/w/cpp/language/enum) and the correct identification would become `NEST::INTERACTION_TYPE::beta` which is guaranteed to be unique.

mszydagis, 2019-08-01:
ok, too much work to solve in couple hours before release this morning then (I was hoping for quick fix where answer one or other "more correct")  So, I will leave this issue open and solve this more properly in the next release in a month or two. Thank you, Luke.

commit 30d8b32c (2019-09-18): Nonfunctional changes related to tidying up (#37)

* NEST::beta, Closes #32

* cc->cpp file naming, Closes #7
closed 2019-09-18 by mszydagis