NESTCollaboration/nest #40: Normal approximation for binomial can be inaccurate

JelleAalbers, 2019-11-04:
When NEST draws binomial random numbers, it currently uses a Gaussian approximation for N > 10 ((https://github.com/NESTCollaboration/nest/blob/master/NEST.cpp#L18). This is inaccurate if p is near 0 or 1. The figure below shows the binomial PMF (solid) and its Gaussian approximation (dashed) for N = 100, for different ps:

[image]

Perhaps it would be possible to implement one of the rules from [here](https://en.wikipedia.org/wiki/Binomial_distribution#Normal_approximation)?

mszydagis, 2019-11-04:
thanks, you even included plot! I will make sure this gets addressed, soon.

mszydagis, 2019-11-27:
@JelleAalbers, This is fixed by https://github.com/NESTCollaboration/nest/commit/11935e3d92912b887deab05de33e18db358e3421  now. Out of the 3-4 options offered by wikipedia I picked the first, testing it for a balance of speed and accuracy, for the NR and ER yields/bands at their most extreme regimes. Thanks for bringing this up. It is a sign of NEST really growing up when users outside the NEST collaboration are starting to make issues. Thank you again!

closed 2019-11-27 by mszydagis