NESTCollaboration/nest #53: New release of NEST

kreczko, 2020-05-15:
Dear all,

I have to admit I am not aware of the NEST versioning policy, so please let me know.

NEST uses the format of <major>.<minor>.<patch> scheme for versioning, which usually means relatively frequent releases (which I am a fan of). However, the last release is already 10 months old - all the new and improved bits are only available in master.

Would it be possible to please release a new version? At LZ we have developers burning to use the latest features and bug fixes:
https://github.com/NESTCollaboration/nest/compare/v2.0.1...master

mszydagis, 2020-05-15:
hi Luke, you must have missed somehow my numerous (verbal only though, not written) announcements at multiple LZ phone meetings the past few weeks: We (including/especially me) are working around the clock towards a new release. A pre-release beta-test version is coming out in 2 weeks for people to do checks with, like compatibility with LZ software, etc.

kreczko, 2020-05-15:
Awesome thanks!

Indeed, I must have missed them, my memory has not been the best lately.

Looking forward to the new version

mszydagis, 2020-05-15:
One thing that is slowing things down, but that you will be appreciate immensely I am sure: a whole suite of validation plots to make sure nothing gets broken by accident during new releases :)

closed 2020-05-22 by mszydagis