NESTCollaboration/nest #54: A nice feature to have?

xxiang4, 2020-05-22:
Two month ago, I forked `nestpy` (source: https://github.com/xxiang4/nestpy) for a LUX project, and added a function `run_NEST_vec()` to my own `nestpy`. This function takes in a list of energies, position as input, and output s1, s2 variables. 

Is it be possible to add a similar function in nest c++ code --- a function that takes in a vector of energies, and positions, and output s1 and s2 variables? Then the python binding will simply mirror the nest version with minimal changes. 

Looping in c++ is often a better idea than looping in python.

tunnell, 2020-05-22:
Have you looked at numpy vectorize?  Should be as fast as C++.  Also, can you push your helper factor back to nestpy for others to use?

Sent from phone

xxiang4, 2020-05-22:
Thanks. To my knowledge, NEST does not have a default function that takes in both energy and position, and output s1, s2. I think numpy vectorization will do it if such a function exist in nest. Otherwise, the vectorization is simply wrapping a python function.

I did fork nestpy to my own repo account. Sorry I don't understand what you mean by "push your helper factor back to nestpy"?

mszydagis, 2020-05-22:
hi Xin, Chris -- it's OK, I will try and handle this today myself, regardless of whether something similar exists in python, since I think this could be really useful, and others on LZ at least will want it, who speak only C++ (like me :) I don't know python)

tunnell, 2020-05-22:
@xxiang4 nice.  I can't comment on adding to C++ side (Matthew can) but I highly recommend you push your nestpy changes in your fork back into the main nestpy.  A few of us started nestpy but it requires people like you to make pull requests of necessary changes to make it a useful tool.  Let me know if you need help knowing how to do pull requests and this process.  This way you can help us back a little if you think that nestpy helped you.  [This is a different issue than what you mentioned, but saw fork and got excited for new contributors]

mszydagis, 2020-05-22:
Closing up since solved with my update referenced above, from 30 minutes ago. However, Xin, as Chris said, as long as there are no private LZ secrets in your code, it is of immense help for you request to push any of your own personal changes to the public NEST repo. Helps the community

xxiang4, 2020-05-23:
Thanks Matthew, no there isn't LZ secret. Just my attempts to make nestpy more useful.

tunnell, 2020-05-29:
Hey Matthew,

I'm going through his changes and it looks like Xin @xxiang4 made a bunch of changes to NEST itself.  Therefore, the following would be required to implement his changes.

Any changes in CPP files have to be changed in this repository since these will be later synced to nestpy.  This would have to be its own pull request to this repository.  These files would be:

-LUX_Run03.hh
-NEST.cpp
-NEST.hh
-RandomGen.cpp
-TestSpectra.cpp
-TestSpectra.hh
-VDetector.cpp
-VDetector.hh
-testNEST.hh
-testNEST.cpp
-analysis.hh 

From here, we could sync nestpy to these changes.  In addition, @xxiang4 would have to for nestpy:

- Update README.md to refer to normal nestpy instead of his fork
- examples/demo_v0.ipynb - move some of these calls also into 'tests' to ensure they work
- bindings.cpp - We just update this to reflect new NEST bindings.

@xxiang4 is this something you'd want to do?  It requires a bit more work to blend it with NEST since currently the state of the code shows that it is your own personal copy and is not yet for general consumption.  

These would have to be done before we can merge: https://github.com/NESTCollaboration/nestpy/pull/30

mszydagis, 2020-06-19:
Dear @xxiang4 Can you confirm you are 100% good now after the last couple git master branch updates/commits? (Extra icing on cake: @grischbieter has tested in python that "runNESTvec" works, and still does after latest changes you requested. CalculateG2 as well) If you can write here and express your approval, I can re-close this issue, and tag stable release (and I don't want to look at the code in any big way for a few months, instead focusing on NEST applications now, so that people on e.g. LZ can use the code without fear of it changing under them so often).

xxiang4, 2020-06-22:
Thanks @mszydagis for the work. It seems to work for me. We can close this issue.

closed 2020-06-22 by mszydagis