NESTCollaboration/nest #64: Zenodo entry for 2.1.0 release shows incorrect date (Jul 18, 2018)

kreczko, 2020-07-02:
On https://zenodo.org/record/3905382#.Xv2SDShKjAQ
[image]

You can see that version 2.1.0 has the same published date as version v2.0.0 and version v0.0.test.

Not sure how this can happen, but it might be worth fixing it for citations.

mszydagis, 2020-07-08:
just making sure @tunnell tagged so he can investigate.

tunnell, 2020-07-08:
Weird.  Though having a look, I noticed a few other things.

I manually fixed this but don't know what happened.

@mszydagis can you make a release within Github release since v2.1.0 is the beta?  https://github.com/NESTCollaboration/nest/releases  In the future, I hope we don't do 'beta' with 'beta' written in the version number, but bump the number then do patch or minor shortly after.  Otherwise, breaks convention (which I think is part of another issue)

[image]

We can manually fix this in the future by adding the date to `.zenodo.json` since Zenodo guessed the date wrong.  I guess we see if it happens again?

mszydagis, 2020-07-08:
hi Chris, I already made a separate 2.1.0 tag and release after beta a few weeks ago, so I am not sure what you mean. I do not know why both don't show on zenodo. See attachment below, if came though. Version 2.1.0beta was tagged and released on May 29, while Version 2.1.0 was tagged and released just 15 days ago by me. I had a separate email about it to the list-host and everything. Only 1 is showing up on zenodo, I am guessing the latter? (hopefully!) Also, on the word beta, I was simply following Jason B's convention from last year on pre-releasing. I didn't know it was not standard or most ideal.

[image]

tunnell, 2020-07-08:
Before we enter the longer discussion, it looks fixed now right?

mszydagis, 2020-07-08:
the date for v2.1.0 of June 23, 2020 does look correct to me now yes, but there is still no entry for v.2.1.0beta from May, the pre-release tag.

tunnell, 2020-07-08:
I'll check

tunnell, 2020-07-08:
I think pre-release doesn't get a Zenodo DOI

mszydagis, 2020-07-08:
but if that were the case, why did Jason's v2.0.1beta, which he had tagged as a pre-release as well, get its own entry?

tunnell, 2020-07-08:
I don't know....  I'll be on the lookout here in the future, but don't know how to investigate further

tunnell, 2020-11-16:
Fixed.  Had to edit .travis.yml to fix.  Was fixing the date.  Now I fixed, and fixed old dates too.

closed 2020-11-16 by tunnell