NESTCollaboration/nest #65: NR mean yields 2.0.1 vs. 2.1.0

jhowl01, 2020-07-30:
Hi all!

If I'm interpreting the release notes of 2.1 correctly, there is no change to the NR best-fit mean yields compared with v0.2.1, but I wanted to confirm this (not sure if this is the proper avenue, sorry if not!).

mszydagis, 2020-07-30:
hi Joey, this venue is totally fine. I can confirm that mean yields have not changed. The width model was very slightly tweaked (no real change, only made equation used more comparable to ER, for apples to apples comparison of models) and a skewness model was added (for NR too, even though trumpeted more for ER) as per: https://arxiv.org/abs/2004.06304  It makes the NR slightly less "drippy" below the mean, but the mean itself of a log(S2) or log(S2/S1) or S2 or S2/S1 band for example should be nearly identical.

closed 2020-07-31 by mszydagis

jhowl01, 2020-07-31:
Thanks for the explanation! This clears it up.