NESTCollaboration/nest #66: Drift Speed non-positive

jarwatson, 2020-07-30:
In the drift velocity calculations, could it throw an error instead of exiting? That way clients can handle the case of errors in the way of their choosing.

mszydagis, 2020-07-30:
hi Reed -- sure, but then how should the code continue executing, when for instance running the standalone testNEST code?? It must assume something for vD in order to able to continue running. Just take absolute value? Can't set to 0.0 otherwise the electrons will drift for eternity, thus halting the sim.

jarwatson, 2020-07-30:
The sim should exit as it does now or the event should be aborted according to what the clients want. It is just that halting the program makes the cause harder to find. It should be up to the clients to sanitize inputs, but at least having a line of code that says "uncaught exception on line XYZ:" would be helpful. 
An uncaught exception exits the program anyway, so from the perspective of testNEST nothing would be altered. It could be something like this: 
```
//testNEST.cpp
try{
  GetDriftVelocity(...);
}catch(std::exception& e){
   std::cerr << "testNEST: Drift Speed too low, check the electric field." << std::endl;
   std:: cerr<< e.what() << std::endl;
  exit(0); // exit just like before.
}
```

mszydagis, 2020-07-31:
Can you tell me an example of an exact temperature in Kelvin, pressure in bar, and drift electric field in V/cm that lead to the issue, so I can dig deeper?

jarwatson, 2020-07-31:
So the particular issue was photoionization spawning electrons in a bad area. In theory, the call to exit is telling us that something worthy of diagnosis is going wrong. However, if something like this is happening there might be some utility to simply killing either that one electron or just that event, rather than cancelling an entire job. Therefore, a throwing an exception, and letting the client(BACCARAT, testNEST) figure out how to recover from this might be more flexible.

In my case, it's insufficient to know that the field is too low, I need to know where the event occurred. Exiting out of the program precludes printing out warning messages when this happens.

jarwatson, 2020-07-31:
Closing this issue. After further consideration it's should be the job of the client to make sure that you're actually in Xenon before calling NEST.

tunnell, 2020-07-31:
I think sanity checks within the function for drift speed would make most sense, no?

Sent from phone

mszydagis, 2020-07-31:
yes, but there need to be more. Currently only checks E-field. Will add: T, P, rho, and/or material.

mszydagis, 2020-08-01:
@jarwatson  How does https://github.com/NESTCollaboration/nest/commit/389e393496eea8e3c3869b786877aa0273f3cf59  strike you? I really went overboard fixing up it I think haha

jarwatson, 2020-08-01:
That looks good to me. If you want to `throw` an exception instead of returning a small value, that would also work.

mszydagis, 2020-08-01:
I actually don't use throw (and catch, try, etc.) anywhere in NEST right now, just wasn't how we wrote it in general couple years ago. I like the ability to keep the code running using a default speed (with the user warned of its value of course, so that they know what sort of drift times to expect). I am going to close this issue up. Thanks Reed.

closed 2020-08-01 by mszydagis