NESTCollaboration/nest #72: Copy operator in `NESTcalc`

riffard, 2020-10-26:
There are a lot of call of the vector copy operators in `NESTcalc`.

Example, here:
```C++
NESTresult NESTcalc::FullCalculation(INTERACTION_TYPE species, double energy,
                                     double density, double dfield, double A,
                                     double Z,
                                     vector<double> NuisParam /*={11.,1.1,0.0480,-0.0533,12.6,0.3,2.,0.3,2.,0.5,1.,1.}*/,
				     vector<double> FreeParam /*={1.,1.,0.1,0.5,0.19,2.25}*/,
                                     bool do_times /*=true*/) 
```
the suggested modification are:
```C++
NESTresult NESTcalc::FullCalculation(INTERACTION_TYPE species, double energy,
                                     double density, double dfield, double A,
                                     double Z,
                                     vector<double>& NuisParam /*={11.,1.1,0.0480,-0.0533,12.6,0.3,2.,0.3,2.,0.5,1.,1.}*/,
				     vector<double>& FreeParam /*={1.,1.,0.1,0.5,0.19,2.25}*/,
                                     bool do_times /*=true*/) 
```
There is the same issue here:
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L38
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L134
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L173
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L402
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L457
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L659
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L1413
https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L1813
I may missed some...

If you want to prevent the vector to be modified, in the function, declare it as const like that:
```c++
const vector<double>& NuisParam
```

mszydagis, 2020-10-26:
hi @riffard I am struggling with the first half of your issue, because you repeated the same identical code twice: I do not see any modifications.

riffard, 2020-10-26:
@mszydagis the modification is: 
```c++
vector<double> NuisParam
```
vs 
```c++
vector<double> &NuisParam
```
but in this context, you can do:
```c++
const vector<double> &NuisParam
```

mszydagis, 2020-10-26:
I feel so stupid, of course! I missed the extra &'s you added, since so subtle.

closed 2020-10-27 by mszydagis