NESTCollaboration/nest #74: Travis takes forever suddenly.

mszydagis, 2020-10-27:
@tunnell  One of the Travis checks is ~instantaneous as normal, but recently (I do not unfortunately recall which exact revision, but it was within the last week during Oct.) the second takes an eternity to complete. See attached screen shot.
[image]

mszydagis, 2020-11-08:
In addition sometimes it only does one of these checks, Chris, and not consistent as to which one it picks (Branch versus travis-ci/push)

tunnell, 2020-11-09:
@mszydagis I see it taking on the order of a minute to complete.  It could just be that sometimes there isn't enough free virtual machines for us to test on.  How long are you waiting?  We can always pay I guess if it gets problematic, but I'm hoping it's just temporary.  The pandemic affects things in weird ways.

[image]

mszydagis, 2020-11-09:
hi Chris, oh not suggesting we pay, especially since NEST is still an unfunded/"informal" collaboration, but it has been taking hours for me, sometimes even days (with overnight not enough for finishing!) I see a semi-permanent yellow circle in place of the green check-mark. But also, note my second message: it will randomly do one check but not the other. I have not discerned any pattern yet to either of these problems. They don't seem to have any rhyme or reason to them. However, I can tell you I have not see only minutes for several weeks at least now.

tunnell, 2020-11-09:
There is also a button to restart a build FYI if this happen

mszydagis, 2020-11-09:
Ah, you will have to show me where that is then. Because I figured something like that had to exist, but for the life of me I could not find it.

tunnell, 2020-11-10:
I looked around, and I think they've just been running slower than usual recently.  The settings in `.travis.yml` seem fine

mszydagis, 2020-11-10:
OK. But what about there being 2 checks sometimes, and sometimes 1 (like in the most recent tagged pre-release)?

mszydagis, 2020-11-15:
hi @tunnell  If you can comment on the 2 checks versus 1 (seems stuck permanently on 1 again lately) then I can close this issue (I do love the fact that git has a label like "wont fix" haha)

tunnell, 2020-11-16:
I think one is for tag and one is for branch?

mszydagis, 2020-11-16:
Well, I don't think that actually follows the pattern. Bottom line is I think neither of us really knows haha, and we cannot do anything about it, especially we have the free version of Travis CI. I'll just give up on understanding the rhyme or reason of this, at least for now.

closed 2020-11-16 by mszydagis