NESTCollaboration/nest #77: Error while installing nestpy

nikita-0209, 2020-11-13:
`pip install nestpy` leads to

```
Collecting nestpy
  Using cached https://files.pythonhosted.org/packages/ed/82/d14b68959d7cc98838267601e8ff9bd9716be0d9d1f549754c1b13d4108f/nestpy-1.3.3.tar.gz
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

tunnell, 2020-11-19:
@nikita-0209 can you post this here?

https://github.com/NESTCollaboration/nestpy/issues

Also, without verbose information on your system or which version of nestpy you want to install, it's hard to give any advice.  See e.g. https://www.chiark.greenend.org.uk/~sgtatham/bugs.html to understand what you have to provide for us better to help.  For instance, did you try an older version?  `pip install nestpy==1.3.1`

tunnell, 2020-11-19:
@sophiaandaloro