NESTCollaboration/nest #90: Using fixed width integer types

riffard, 2021-02-22:
Modern C++ requires to use of the fixed-width integer types that were introduced in C++11. It's recommended to use `int64_t` instead of `long int`. The header is `cstdint`: [documentation](https://en.cppreference.com/w/cpp/types/integer).

tunnell, 2021-02-23:
These are good suggestions.  If it's quick, you have a better feeling for C++11 than we do, so you're welcome to try your hand at a PR that we can test to make sure things function fine.  Right now NEST is maintaining more than developing, so this will happen but probably slowly.

mszydagis, 2021-02-23:
I have fixed this today, didn't take long thanks to grep, Quentin and Chris -- Will be uploading later tonight, combining with the other minor issues. I even went one step further and changed all unsigned long or unsigned long int's to uint64_t

closed 2021-02-23 by mszydagis