NESTCollaboration/nest #94: segfault after commits 476c376 and further

Aspelene, 2021-03-02:
Hi all,
I've encountered a problem after new commits: I have segmentation fault for any model (for all fields and energy ranges, pure build w/o ROOT/Geant) + warnings during compilation about long long int format (only with output part (sprintf), not in "main" code itself). Tried to change uint_64t and int64_t to old unsigned long /long (according to logs), but it didn't help.
Please see compilation log with warnings attached. OS is Ubuntu 20.10, CXX compiler is GNU 10.2.0.
[log_nest.txt](https://github.com/NESTCollaboration/nest/files/6071003/log_nest.txt)

riffard, 2021-03-02:
Hello @Aspelene, 

Could you send me the log of the code execution? This would help us to debug what going on. 
What is the exact command to are executing?
`int64_t` are not responsible for this kind of issues.

Thanks,
Quentin

Aspelene, 2021-03-03:
Hi Quentin!
Sure, please see below (I'm using Ar models (changed xenon numbers to argon in NEST.hh and TestSpectra.hh), that's why there is argon warning message):
./execNEST 25000 beta 9.4 9.4 7 -1 
*** Detector definition message ***
You are currently using the LUX Run03 template detector.
WARNING: Argon is currently only in alpha testing mode!! Many features copied over from Xenon wholesale still. Use models at your own risk.

Segmentation fault (core dumped)

The result is the same with any field/energy range/model.
Best, Kate.

mszydagis, 2021-03-03:
hi Kate, that could be the problem then, since I moved the Ar model around and I easily could have broken something (hence my earlier request for you to check! :) Can you run just the Xe model in a pure unchanged piece of NEST, and confirm you are still having problems? That will tell us if the problem is general or not. You may have already done this ("The result is the same with any field/energy range/model." you said) but I want to make sure.

Aspelene, 2021-03-03:
Hi Matthew,
Yep, the problem was only with argon and I think I found a solution - the problem was in definition of NuisParam for argon (values were assigned via NuisParam[i], not via push_back). When vector methods (push_back) are used everything is working fine. Will add a commit right now.

mszydagis, 2021-03-03:
Ah, stupid me, this was my fault then! But, because only LAr affected and nothing else, and we have not officially released LAr model since ER model still has serious problems, I won't release an official patch. You can make a merge request of course though, and we will put on the bleeding edge. It will be included in the next release in April

closed 2021-03-03 by mszydagis