NESTCollaboration/nest #97: Too verbose Kr83m deltaT range warning

sophiaandaloro, 2021-03-08:
The new Kr83m deltaT warning alerts users of deltaT being out of bounds but should only be called once so that those simulating many kr83m yields do not get too many warnings. 
Relevant PR: #92 
Code: 
The proposed solution is as follows (so that it appears once, whether or not you are in a loop or vectorizing.) 
Replace the new code with something to the extent of 
```
bool reported_low_deltaT = false; // determining where this goes will be the trick but I have a few ideas. 

 GetYieldKr83m(**args) {
// do a bunch of stuff in between here. 
    if (deltaT_ns < 100 && energy < 41.5 && reported_low_deltaT == false)  {
       reported_low_deltaT = true;
       cerr << "\tWARNING! Past Kr83m model fit validity region. Details: "
       << " deltaT_ns is <100 ns and your input energy is either 32.1 or 9.4 keV. "
       << " Data for separated Kr83m decays does not yet exist for deltaT_ns <100 ns. "
       << " 9.4 & 32.1 keV yields are still summed to physically accurate result, but individually will be nonsensical." << endl;
     }
}
```

This ideally means the warning only outputs once as long as we initialize the reported boolean outside of the function (inside would be a slightly different story, but also an option.) 

Do not have time to fix this this week probably but as we discussed this should be an open issue so we don't forget.

sophiaandaloro, 2021-03-08:
@vvelan @mszydagis Here's the Kr83m warning issue, almost forgot it!

vvelan, 2021-03-08:
That strategy looks good to me!

closed 2021-05-22 by mszydagis