NESTCollaboration/nest #99: convert all of the C-style arrays

mszydagis, 2021-04-09:
into std::array or vectors, everywhere in the code (might be big job)

kreczko, 2021-10-20:
Note: Now that you are using `C++17, you might also want to look at `constexpr` for all the `const` vectors/arrays (or lookup tables):

https://joelfilho.com/blog/2020/compile_time_lookup_tables_in_cpp/

infophysics, 2022-04-24:
Apparently "constexpr std::vector<T>" is now standard in c++20, but not for earlier versions.

mszydagis, 2022-04-24:
So, we are only at C++17 in NEST, and even then we only just recently upgraded from using the C++11 standard (and for good reasons: to maintain compatibility with XENON's code, especially in python, so we canNOT upgrade, at least not at this time). The code works, and runs fast, so let's just let sleeping dogs lie on this one :-) I will leave the issue open though.