NESTCollaboration/nest #101: Kr83m model for deltaT< 100 ns

terliuk, 2021-05-03:
I was recently trying to produce two separate Kr83m lines and found that there are couple of peculiar features for times below 100 ns. One of them is the behavior of electron and photon yields for times around 50 ns, where total number of electrons for two lines drops from ~1300 to 700 in just 10 ns time difference. Is there a reason why it looks so or is it a glitch of the underlying model? 
 
[image]

Also, I was expecting that total number of electrons + photons stays the same regardless of decay time, but it seems to have a feature at 50-100 ns. 

[image]

I understand that two lines should not be trusted individually at such time scales, but it would still be nice to have a bit more realistic extrapolation into that region, as different experiments might be sensitive to double peak photon timing at rather short time scales. I have created plot for Kr83m with 30 ns time difference and an "equivalent" 2 gamma rays with the same energies to demonstrate this potential effect. 

[image]
[image]

All of the plots are made with NESTpy 1.4.8. 
Great thanks for help!

mszydagis, 2021-05-03:
hi @terliuk there is a method to the madness here! @grischbieter and @vvelan can explain: despite some unphysical features this is literally the best we could do for the Kr83m model, after years of effort: this is because of the extremely contradictory data. The model won't be updated any time soon.

terliuk, 2021-05-03:
Great thanks for clarification! One small question - should we then be concerned about number of photons + electrons being not stable at different delta T?

mszydagis, 2021-05-03:
Ah, for that Greg or Vetri will have to comment -- but I do think no experiment can get below 50-100 ns at least because of the exponential triplet time constant of 25-30 ns. The S1s really become inseparable.

grischbieter, 2021-05-04:
Hi @terliuk I'll just add that, while I don't think you should be concerned with the behavior for T < 100 ns, I believe results in this regime should only be loosely considered.  As @mszydagis mentioned, this behavior is due to balancing very contradictory data, thus one should expect that the NEST models haven't isolated the "true" behavior of course.  The main data set used to guide the newest 83mKr model is the ~recent PIXeY 83mKr analysis https://arxiv.org/pdf/1911.03999.pdf (as it is the most complete collection of times and drift fields), while balancing agreement with data from Xurich II, LUX, XENON100/1T, PandaX, and others. 

I'll note specifically, that the time-dependence seen in https://arxiv.org/abs/1303.6891 is similar to that shown by PIXeY, yet the field-dependence is greater in the PIXeY data. Additionally, the time-dependence in https://arxiv.org/abs/1209.3658 by Aprile et al. is very different, where the yields continue to decrease even as the time-separation approaches 1 us.  After working out the model to balance the measured time-dependence, the unmeasured behavior (T < ~100ns) was used to match the "merged" Kr83m yields from the large TPC experiments (LUX/XENON/PandaX). 

Hopefully this helps, and thanks for the questions. Let me know if we can help further.

sophiafarrell, 2022-04-22:
@mszydagis I think this should be reopened. The way the Kr83m yields function is written means that the random sampling will sometimes result in a nonphysical value for yields (not constant sum) for deltaT<150 ns, even for 41.5 it seems by the plots above. Whether or not the user has some statement to remove 9.4/32.1 keV separated events from their simulations with low deltaT, this still is going to trigger a problem for 41.5 keV and cause us to have an abnormally high total yields value when this is just an artifact from fitting, as I understand it. 

I'm not 100% sure what the solution is, but I don't think the Kr83m model is correct below 200 ns, as it produces total yields/keV that are not constant.

mszydagis, 2022-04-22:
I agree, but the issue is that no one is available to work on it. Are you volunteering?? :) Greg R. and Vetri are thesis-writing.

sophiafarrell, 2022-04-22:
Thanks for reopening! Even though no one can work on it right now, keeping existing issues open means that people can work on it when they find the time (and it stays relevant and prioritized), so I would advocate that we keep this issue open until fixed. It allows others to fix it if they stumble across it and have a solution (has been true for me on nestpy issues at least). Also, we can reference that we're aware of it and there is an open issue when it comes up (this particular issue comes up semi-regularly), and close any duplicate issues in favor of this one that may arise.

mszydagis, 2023-06-23:
@terliuk  @sophiafarrell  @grischbieter @vvelan  Finally fixed this once and for all (I hope!!)
9.4 keV yields still go unphysical (above 73 quanta/keV total) at very low S1b-S1a separation times and low E-fields so Nq/E or W is still *not* fixed, but this is necessary to explain 41.5 keV "merged" yields, and should at least be smooth now. I have re-checked agreement with all available data and it's looking good. I have compromised on the time dependence (some experiments now claim none!) by moderating the dependence on S1a-S1b separation seen by PIXeY, ignoring the extreme low-time jumps (mathematically impossible by convservation of energy, always were, as they would require >100 photons/keV! I am guessing PIXeY had pulse area overlaps due to the S1 recombination time especially at low fields) and ignoring "humps," going only for smooth, monotonic modeling. See:
https://github.com/NESTCollaboration/nest/commit/1cc7905bed616f98319b8b95a8b5518095547032 

Any disagreement you see with data below: fixing it would BREAK ageement with a *different* paper (not all featured below, as there are sooo many now..)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2023-06-23 by mszydagis