NESTCollaboration/nest #102: Density argument not used in Drift Velocity calculation

sophiaandaloro, 2021-05-12:
Here is the definition of the `GetDriftVelocity_Liquid` function: https://github.com/NESTCollaboration/nest/blob/c622b0937b1971c217e1659ffb2cee87135bf393/src/NEST.cpp#L1588

To my knowledge, `Density` is not called anywhere else in the function. Can we remove it?

mszydagis, 2021-05-12:
I think the idea was to use it eventually, so it is vestigial. Temperature does not describe the liquid alone as we currently use, so if we got rid of density we might have to add pressure one day.

sophiaandaloro, 2021-05-12:
Thanks for the fast clarification! I am not sure what the C++ best-practices is for vestigial parameters. My python-idea would be to set a default (so users don't have to specify it, thus don't assume it is important if they are not experts) and comment in the header file this point. Does that sound c++ worthy? Or any other suggestion? (Can loop in Chris too for his help if we need a second opinion since he typically knows the best way to document things...)

mszydagis, 2021-05-12:
I concur with you! Instead of deleting, give it a default (e.g., 2.9 sounds good for liquid :) The units are g/cm^3. It is getting to be that time again (over 1 month since last NEST patch) to start fixing up the code in preparation for a new release.

sophiaandaloro, 2021-05-16:
Fixed by #103

closed 2021-05-16 by sophiaandaloro