NESTCollaboration/nest #106: S1 efficiency depends on nHits

sophiaandaloro, 2021-05-18:
At request, turning this question into an issue. 
It seems that according to these lines:
https://github.com/NESTCollaboration/nest/blob/3322903b7cd2439c349418ad25e6caeaa820acfe/src/NEST.cpp#L804-L808

There is an "effective" SPE efficiency that depends on the number of hits, `nHits`. This comes into the S1 Pulse area at this timingless S1 case: 
https://github.com/NESTCollaboration/nest/blob/3322903b7cd2439c349418ad25e6caeaa820acfe/src/NEST.cpp#L887-L889

The question that @robertsjames and I have is whether or not this "effective" efficiency is derived analytically, or fitted (where did it come from in functional form?) and if we can re-parameterize it in terms of something other than nHits. For example, in an ideal case, we would like for `eff` to be dependent instead on the number of photoelectrons produced or detected, such as this variable: 
https://github.com/NESTCollaboration/nest/blob/3322903b7cd2439c349418ad25e6caeaa820acfe/src/NEST.cpp#L886

This `eff` is important because it has a ~few% effect for detectors with an SPE efficiency of <99%, which affects energy reconstruction, simulations, etc. So we want to get this right on our end, but also ensure that NEST has it right, too!

mszydagis, 2021-05-20:
This has been changed, to where efficiency depends on Nphe. While that makes it slightly circular, the other issue (#107) alleviates this, by better distinguishing between Nphe and Nphe_det for *detected*. So, I am closing this too. Should be fully solved. Making the efficiency depend on the pulse area in PHE (or PE, same thing, different collaborations...) is actually better, since the 2-PE effect gives you an efficiency boost, like the LZ and LUX papers on that topic say (low-mass DM). Lastly, I added a comment in the code that justifies the functional form. It's super trivial, just a linear approximation.

closed 2021-05-20 by mszydagis

sophiaandaloro, 2021-05-20:
(Just referencing the commit here so anyone searching this can see the change in full form https://github.com/NESTCollaboration/nest/commit/9363b9689d9fd7afaeef8cbf4178a0baeb2bcba9)