NESTCollaboration/nest #129: RecombOmegaER obscurities

sophiafarrell, 2021-09-24:
Perhaps this will get resolved soon, but it seems a bit unclear whether the commented out regions in RecombOmegaER should all become uncommented to match the BetaGR model (new beta model).  
In either case (no matter how we define new/old beta models), we should probably be very clear about this, since this will significantly affect any ER calculations. 
**My suggestion would be in the next release to:** 
have an argument "use_old_beta=False" or something, to indicate we aren't using the old beta model, which should be the default. If this is false, nothing happens. If true, we reset some of the hardcoded fit numbers in the function to the old values. 
https://github.com/NESTCollaboration/nest/blob/c7701a87df990c1ed3e10bef78cc4fd9abfa81d6/src/NEST.cpp#L179-L194

mszydagis, 2021-09-27:
hi Sophia, I worry that both the analysis header file and the detector files have way too many options these days, so instead of adding another flag, I just went through RecombOmegaER as well as other places in NEST.cpp and I made the comments extremely clear. I hope that helps. I am uploading my changes shortly, so closing the issue. Beyond the XENON1T 82 V/cm result, there were a couple other reasons we had to delay this model change, but they were not all great reasons, so this change is very long overdue, since it incorporates years-old ER calibrations now. Looking again at the plots for the paper, this model works so much better (except for possibly at 50 and 180 V/cm slightly worse than old model) that I think we should discourage anyone from using the old, even though as you say this is a big change. The code comments will preserve the original model perfectly, which is not getting deleted in the foreseeable future, and it only takes a few seconds to restore it, even though that does mean code hacking instead of flag setting.

closed 2021-09-27 by mszydagis

sophiafarrell, 2021-09-28:
Hey Matthew, 
The documentation certainly does help! However, this still causes issues for people installing nestpy from source. I mean, it will be not *catastrophic* as long as we change to the new model the new parameters by default, but for nestpy, people will have to know how to compile from source (which to be honest, I've even forgotten how to do in the last 3 months while away, since it's a bit technical) So, just a forewarning that this may cause some issues on the nestpy side.
On my end, I can at least ensure the python functions have commented documentation explaining this, and that will hopefully raise a lot less issues. But I would say in general, this may not be best practices (for python at least).
In the future, we might think about how to handle these things better, just because I know that NEST can get more and more complex as time goes on and for the collabs using python, this ends up being a TON of front-end user work to figure out how to recompile C++-python bindings. But, I digress. The improved documentation is enough for this tag, probably :) Thanks!