NESTCollaboration/nest #132: Units for detector parameters

tunnell, 2021-10-05:
A XENON person just (reasonably) asked why there aren't units in the detector files:

https://github.com/NESTCollaboration/nest/blob/master/include/Detectors/LUX_Run03.hh#L29

Could somebody add?  Just adding here so don't forget.

mszydagis, 2021-10-05:
Hi Chris, there are units in the original detector file DetectorExample_XENON10.hh, and they've been there for years, since the beginning of NEST essentially. Seems kind of painful to repeat for each detector. However, I can add to LUX Run03 since it is the default.

mszydagis, 2021-10-05:
@tunnell , addressed in https://github.com/NESTCollaboration/nest/commit/178c5e2031053d22b40119100f78a31e3baaf967

closed 2021-10-05 by mszydagis