NESTCollaboration/nest #139: Kr83m deltaT_ns

HenningSE, 2021-11-02:
The `GetYieldsKr83m` function has the argument `deltaT_ns` (defined here: https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L617-L618). When this function is called in `GetYields` (here: https://github.com/NESTCollaboration/nest/blob/master/src/NEST.cpp#L829-L830) the argument is not used. So `deltaT_ns` is always set to the default value of  `-999` which gives this warning: `WARNING! Past Kr83m model fit validity region. Details:  deltaT_ns is <100 ns and your input energy is either 32.1 or 9.4 keV.  Data for separated Kr83m decays does not yet exist for deltaT_ns <100 ns.  9.4 & 32.1 keV yields are still summed to physically accurate result, but individually will be nonsensical.`

mszydagis, 2021-11-02:
hey @sophiafarrell can you figure out what is going wrong here? Thanks

sophiafarrell, 2021-11-02:
Hi @HenningSE thanks for opening the issue. This warning is only for energy < 41.5 (saying you can separate the peaks). I thought you were having issues with the 41.5 keV? 
In any case, the deltaT_ns is sampled for any energy state we're looking at in that yields function, if you pass -999. So you're getting this warning in cases where you're looking at 32.1 or 9.4 keV decay modes. If you would like to set it so the maxTimeSeparation is over 100 ns (and in the region of validity), then just pass something >100 to massNum argument in GetYields, or use the binding to GetYieldKr83m directly. 
Let me know if that is not what you needed.

sophiafarrell, 2021-11-02:
So the proposal we have to make things more flexible for the user is: 

https://github.com/NESTCollaboration/nest/blob/7cd10e2b16b2926cea1322cff44d35251066762b/src/NEST.cpp#L617-L618

Add in an argument like `massNum` here, but instead it'll be `atomNum`, that serves as the stand-in for `deltaT_ns` such that the user can hard-code from `GetYields`. 
That said, the default should still be -999, or I suppose maybe @mszydagis or another C++ knowledgable person can help me figure out how we can ensure we don't accidentally pass in the wrong default parameter. Basically, if we do just throw in `atomNum`, and you use GetYields, without filling in a value for this argument, will it still default to `-999`? Thanks and sorry for the lack of C++ arg knowledge!

mszydagis, 2021-11-03:
hi @HenningSE  I am trying to fix the issue myself today, but am very confused: why can't you just call "YieldResult NESTcalc::GetYields(INTERACTION_TYPE species, double energy, double density, double dfield, double massNum,
                                double atomNum, const std::vector<double> &NuisParam" (line 800) directly, with the species specified as Kr83m and the "massNum" variable re-used as your maxTime? I am not able to reproduce the problem.

HenningSE, 2021-11-08:
Hi @mszydagis, thanks for having a look. My problem is that `GetYieldKr83m` has the option to get `deltaT_ns` as input but one can not pass this value if one is using the `GetYields` function. In principle I could directly use `GetYieldKr83m` but I'm also simulating other interaction types and `GetYields` is quite convenient as it acts as wrapper function for all cases NEST can handle.

mszydagis, 2021-11-08:
hi @HenningSE , this may not fix everything then for you, but based on your git issue here I overhauled the Kr83m code this weekend to make it more flexible, comprehensive, and user-friendly: https://github.com/NESTCollaboration/nest/commit/78e6a5fcbaa51733466c3a18d00cfcc44479204f  followed by https://github.com/NESTCollaboration/nest/commit/a7948e7a6e881abf7b7dd9f6e281664df20685c0  to fix a few mistakes I made still.

sophiafarrell, 2021-11-08:
@HenningSE does this not actually fix your original issue, though? Because now that we're using massNum and atomNum as placeholders for min and max time separation, if you set them equal, you can sample exactly a single time. Those arguments are accessible through `GetYields`.

mszydagis, 2021-11-08:
@sophiafarrell  I am not 100% certain that is correct, but of course @HenningSE  will have to comment and correct me if I am mistaken. Trust me I would of course be THRILLED if we're finally done haha, since this was a lot of work to fix up all of this, but my understanding is that Henning's concern is that atomic number and mass number being co-opted breaks GetYields for NR and for ions. However, let me explain why I did it this way: we are already adding so many additional variables to GetYields. Many functions in NEST have way too many arguments already. In addition to making the code look long, and be unwieldy, they also may slow it down, and make it less flexible, repeatable, copyable into other applications, etc. etc. etc. Thus, since Kr83m is a ONE massive exception that requires a couple more numbers, I am a  huge fan of "polymorphic" arguments, as long as they're well documented of course (I've updated the README file for one). Therefore, I absolutely cannot add 2 more numbers to GetYields, which are >99% inapplicable (doesn't apply to NR, "normal" ER, alphas, etc. etc. etc.) as it is already way too long and unwieldy. So, I do hope that is not what the original git issue was asking for me/us to do?? This is my compromise: just don't call the same GetYields for Kr83m and non-Kr83m work, just separate them, to not "break" NR and ions. You can still as Sophia mentioned always call GetYieldKr83m directly, I am unsure why that is less good?? (Kr83m doesn't need mass number of course since it is not Xe-Xe NR or non-Xe nuclei moving through the Xe such as alpha particles -- and atomic number not needed either, because in LAr for instance we just use the beta model, as there is not enough evidence of Kr83m "weirdness" in Argon; this is why I overwrote those two parameters...)

HenningSE, 2021-11-10:
Hi @mszydagis, Hi @sophiafarrell, I'm pretty sure that this solves my problem. Thank you!

closed 2021-11-10 by HenningSE

mszydagis, 2021-11-10:
@HenningSE  Thank you! I wanted to ask though if you were involved in this paper? (https://arxiv.org/pdf/2109.13735.pdf) One thing I wanted to share with you: NEST does appear to match your Kr83m data rather well, but the interpretation was kind of tricky of my part. Usually, L0 means the light yield of individual peaks at 0 V/cm but I think you used 1/15.6 eV, same value as your Q0? In the future would it be possible to report Ly and Qy in absolute photons and electrons/keV? In that way your data is directly comparable to NEST and able to inform it. You could pull out your g1 and g2 using a Doke plot, which you already have (Figure 15) But even with absolute yields, I wanted to let you know there is too much contradictory Kr83m data out there already, between Manalaysay 2013 the seminal paper, already in your plot, but also Baudis 2020 which you did not include (and an earlier Zurich paper) plus also PIXeY which you also did not include a comparison to in your current paper version (apologies if I am mistaken and this is NOT your paper??? arXiv is down this morning and I cannot re-open it to look at the author list)

Interpreting your L0 and Q0 as both in reference to 15.6 eV (which by the way is not related to the 11.5 eV, that comparison should be 13.7 eV with 11.5 eV. The 15.6 eV you quote is W_i NOT W they are not the same things, as explained in the 2011 NEST paper in JINST: 15.6 can be derived from 13.7, or 11.5, eV using the exciton-ion ratio: https://iopscience.iop.org/article/10.1088/1748-0221/6/10/P10002) Then we get a pretty decent agreement, with the exception of 9.4 keV, but that we cannot fix, because we need to also follow the PIXeY, Zurich I & II, 2013, LUX, XENON100, and PandaX published Kr83m data points. NEST was already averaging over many contradictions (the dual points for 9.4 keV are using 300-2,000 ns or just 2,000 ns, flat yield, points -- weren't sure what to use, and also diverging from L0 15.6 interpretation to force it to look better, because indeed we do have a contradiction for 9.4). I hope this is helpful, and if I'm mistaken in contacting you about this, please could you forward my comments to the authors of the paper in question? My last comment is that at low E-fields the existing NEST is extremely powerful already and on top of nearly all data.

[image]