NESTCollaboration/nest #155: Missing references for longitudinal diffusion at lower drift field

YaniBion, 2022-05-24:
I’m trying to compare the longitudinal diffusion values from our detector to the ones in the literature + the ones modelled by NEST.

If I plot the function used in NEST I get something like this:

[image]

Which goes to zero at lower fields (<100 V/cm). 

I'm failing to find where are the references for low field. It is written: 
'  if (!highFieldModel) {
    output = 345.92 * pow(dfield, -0.47880) *
             exp(-81.3230 / dfield);  // fit to Aprile & Doke review
                                      // paper and to arXiv:1102.2865;
                                      // plus, LUX Run02+03 '
The Aprile and Doke review only contains data for high fields:
[image]

Peter Sorensen's measurement is around 700-800 keV. So I’m guessing lower fields have to come from Lux data, yet I cannot . In this paper from 2019, the values go higher, and not to zero, at lower fields: https://doi.org/10.1016/J.NIMA.2020.163965.
[image]
Same in Erik Hogenbirk's thesis.

Could someone throw some light on this?

mszydagis, 2022-05-24:
Absolutely: I never imagined anyone would need the value at a such a low drift E-field so I just over-fit this to one LUX data point and forgot about this years ago. The low-field data is based on LUX's non-science run on the surface called Run02 (famed Run03 was the first WIMP search run, followed by Run04). The field was ~62 V/cm. The diffusion constant in NEST comes from fitting LUX data that was never published actually, but can be found in the PhD thesis of Brown university graduate student Jeremy Chapman (PhD advisor Rick Gaitskell). I am copying the plot below (black is raw data, blue is the binned and fit data, while gold is NEST.) However, this may be a fluke point (our field was perhaps non-uniform for example and we never went back to that data, since it was only a technically/engineering run, not even underground!) In addition, instead of going to 0 perhaps it flattens or maxes out, as the plots you shared show. I just picked a nice analytical function at the time, when data was very lacking. Therefore, thank you for bringing this to my attention -- we will fix this serious mistake and make this part of the June release of NEST next month (v2.3.8) See below for the plot though that answers your current question most immediately. Lastly, there is a solution right now for you: LZ student Ryan Linehan noticed a similar problem as you did a few months ago but for high field. He implemented a fit to the "Boyle" theoretical model, and this is now optional within NEST even though it is not the default. While I look at refitting the *global* data including all the latest from Nikhef and Zurich and XENON et al., please consider using the Boyle model. @relineha may be able to help you. I will keep this issue open until we put in a permanent fix in a couple of weeks.

[image]

mszydagis, 2022-05-31:
hi @YaniBion  I just wanted to check in with you on this git issue.

YaniBion, 2022-06-01:
Hi Matthew, sorry for my long silence (I'm in the frenzied period of writing my thesis). I have checked the thesis, and indeed it would like the diffusion goes to zero, although it contradicts Nikhef, Zurich, and the paper I attached before. I would say, it is very unlikely that the diffusion really vanishes at low fields.

mszydagis, 2022-06-09:
hi @YaniBion  I finally circled back around to addressing this. The June release (tonight or at latest tomorrow) will have a new model for D_L. Please see the attached plot below. Dashes lines are all NEST. Orange is D_L, and the one that goes too low at both low and high E-fields is the OLD model. The new one curves up higher at low fields (also goes slightly higher at high fields, to hit Shibamura). Dashed yellow is D_T, and that will remain UNCHANGED. As you can see, NEST already does a great job (the past ~3+ years) of matching both Doke and EXO, and splitting the difference where they are contradictory (around 1,000 V/cm, although we will probably never go that high again in LXe TPCs in the future...) I have put in the Boyle model in solid (brown for D_L and cyan for D_T). As you see, it unfortunately doesn't make any sense, even though this is the "official" theory of diffusion in noble liquids!! It doesn't match ANYONE's data! So, I guess we will have to stick with (semi-)empirical models then still. I tried to split the difference between the contradictory data at low fields, while at high I force NEST to now go through the unpublished green triangles. Note that NEST's D_L is still systematically low across the board, but it is fit to Peter Sorensen's seminal paper on this (cited in the code, always was) and in that way forced to agree with both XENON10 (730 V/cm) and XENON100 (530) S2 width data very precisely (also, LUX Run03 at ~180 V/cm). I do think the contradictions may be due to slightly differing definitions between NEST and Hogenbirk 2018 especially, on "diffusion," as we require an offset as per Peter's paper (basically, even zero drift has some "intrinsic diffusion" due to the energy and time required for thermal electrons to "pop out" of the liquid). The bottom line is: can we reproduce the raw real data on S2 width versus drift time in final analyses, and if we can then we're all good.

[image]

closed 2022-06-09 by mszydagis