NESTCollaboration/nest #190: The distribution of recombination time in photon timing

FaroutYLq, 2025-01-03:
While reading the codes for photon timing, I am a bit confused by [this line](https://github.com/NESTCollaboration/nest/blob/c3c5b8a1361b70aa585ab8e4e58d95e11e790857/src/NEST.cpp#L120). This line suggests that the recombination time is following sort of a power law distribution. 
```cpp
time_ns += tauR * (1.0 / RandomGen::rndm()->rand_uniform() - 1.);
```
However, the [annotation right above](https://github.com/NESTCollaboration/nest/blob/c3c5b8a1361b70aa585ab8e4e58d95e11e790857/src/NEST.cpp#L118-L119) suggests it to be an exponential distribution. The [published paper on NEST pulse shape ](https://arxiv.org/abs/1310.1117) also suggests the same. 

I wonder if I missed something obvious? Thanks in advance!

mszydagis, 2025-01-03:
Hello, the code is correct. The comment says "the recombination time is non-exponential" following the Kubota 1979 paper (
[Kubota 1979 Dynamic behaviour of free electrons in the recombination process in liquid argon krypton and xenon.pdf](https://github.com/user-attachments/files/18296258/Kubota.1979.Dynamic.behaviour.of.free.electrons.in.the.recombination.process.in.liquid.argon.krypton.and.xenon.pdf) Please see equations 3 and 7 in there. 
The 2013 NEST paper is severely out of date unfortunately. This non-exp (power law) time constant is mentioned on page 19 right of https://arxiv.org/pdf/2211.10726v3 (....even if unlike the singlet and triplet decay profiles the recombination one is non-exponential (t^-2) [145]..)

closed 2025-01-03 by mszydagis

FaroutYLq, 2025-01-03:
Thank you Mathew!

FaroutYLq, 2025-01-04:
Hi again. I have followup question. I wonder where does the [analytical form of energy dependency](https://github.com/NESTCollaboration/nest/blob/c3c5b8a1361b70aa585ab8e4e58d95e11e790857/src/NEST.cpp#L94) of `tauR` come from? It doesn't seem to be the magenta curve from Figure 1 in the [reference in comment](https://arxiv.org/pdf/1310.1117). Thanks in advance for your advice.

mszydagis, 2025-01-05:
Yes, it doesn't match because the paper is outdated and there was no time to write a new one specifically on S1 pulse shape; the updated curve follows the real data better. (It has no other justiifcation that that: empirical.)

FaroutYLq, 2025-01-07:
> Hello, the code is correct. The comment says "the recombination time is non-exponential" following the Kubota 1979 paper ( [Kubota 1979 Dynamic behaviour of free electrons in the recombination [...]

Sorry for another question regarding ~t^-2.

- In the [code](https://github.com/NESTCollaboration/nest/blob/c3c5b8a1361b70aa585ab8e4e58d95e11e790857/src/NEST.cpp#L120-L126), the recombination time following ~t^-2 is sampled before sampling singlet/triplet decay.
- In Kubota 1979, it seems that the timing of recombination follows ~t^-1 (Eqn 3), while timing of emitting photon (under slow recombination limit) follows ~t^-2 (Eon 7). 

It seems to me that the referred Kubota picture suggests that we should sample from ~t^-1 in [this line](https://github.com/NESTCollaboration/nest/blob/c3c5b8a1361b70aa585ab8e4e58d95e11e790857/src/NEST.cpp#L120). Did I miss something? 

Thanks a lot for all your answers!

mszydagis, 2025-01-07:
I believe the issue is the difference between intensity, where a square comes in, and raw amplitude. Because of integration an extra factor of t is gained. However, this was coded up many, many years ago by me, so it is possible I made a mistake. You can try different methods to gain a better fit to data.