NESTCollaboration/nest #191: Organic Scintillator Light Yields

ParticleTruthSeeker, 2025-01-19:
Can this code be applied to or extended for use in the ab initio determination of light yield of organic scintilators?

mszydagis, 2025-01-19:
I am sorry, but it cannot. We don't have the time, students, and resources. But, the built-in Geant4 scintillator class should work

closed 2025-01-19 by mszydagis

ParticleTruthSeeker, 2025-01-19:
Thank you for the reply! Concerning GEANT4, I was under the impression that the light yield as a function of de/dx is dependent on some constant that it seems NEST can provide for LAr and Xe through theortical and empirical methods. Could you perhaps point me to the part of the code that performs this function through theory? What tasks might a student be required to complete to extend these calculations for other materials?

My apologies if I misunderstand the nature of the project.

Best Regards

mszydagis, 2025-01-19:
No, NEST cannot provide that because the noble elements don't just have one constant or one function, they are more complicated than organic scintillators. See example our latest paper: https://arxiv.org/abs/2211.10726 So, it's apples and oranges (although, organic scintillators might be easier than Xe or Ar acually: just 1 Birks constant kB as you suggest)

ParticleTruthSeeker, 2025-01-19:
Ok thank you. I will have a read. Hopefully it will lead me on the right track!